OpenCyphal/libcanard #12: Implement drivers for STM32

pavel-kirienko, 2016-02-14:


jschall, 2016-09-20:
@pavel-kirienko I'd like to take this on. Do you have any information on getting started?

pavel-kirienko, 2016-09-21:
@jschall Great. These things may help you with this:

- STM32 driver for Libuavcan: https://github.com/UAVCAN/libuavcan/tree/master/libuavcan_drivers/stm32/driver. It is very complex, but its Libcanard port will be much simpler.
- Documentation for the above: http://uavcan.org/Implementations/Libuavcan/Platforms/STM32/
- STM32 CAN driver for the PX4 UAVCAN bootloader: https://github.com/Zubax/PX4Firmware/blob/px4esc_bootloader/src/drivers/bootloaders/src/arch/stm32/drivers/can/driver.c. I wouldn't recommend using this code as an inspiration since it's extremely special-purpose and non-portable (it was created for the PX4 UAVCAN bootloader which has very strict code size constraints).
- Obviously, existing drivers for other platforms such as AVR, and the official STM32 documentation.

Also please make sure to skim through this https://github.com/UAVCAN/libcanard/blob/master/DESIGN.md.

Hope this helps!

closed 2017-03-30 by pavel-kirienko