OpenCyphal/libcanard #14: Pack Structures

masierra, 2016-02-18:
Structure padding on different compilers can result in CanardRxState being too large. May need to pack them in a portable way.

antoinealb, 2016-02-18:
yes we will need to use the uavcan standard for this. In the meantime something such as captn'proto or protobuf can be used if you dont require compatibility with libuavcan.

pavel-kirienko, 2016-02-18:
No it's about `CanardRxState` being too large. The problem has no relation to the protocol itself.

antoinealb, 2016-02-18:
Oh sorry, I answered too quickly, my bad... That's what I get for answering from my phone :wink:

I don't think there is a standard structure padding but supporting gcc and clang is probably enough, and they both accept `__attribute__`.

pavel-kirienko, 2016-02-18:
Solution (continuing the Skype conversation):

```c
#define CANARD_RX_PAYLOAD_HEAD_SIZE (CANARD_MEM_BLOCK_SIZE - offsetof(CanardRxState, buffer_head))
#define CANARD_BUFFER_BLOCK_DATA_SIZE (CANARD_MEM_BLOCK_SIZE - offsetof(CanardBufferBlock, data))
```

(not tested)

masierra, 2016-02-19:
I tested it on my linux machine and it seems to work. Push the change?

pavel-kirienko, 2016-02-19:
Sure.

closed 2016-02-20 by pavel-kirienko