OpenCyphal/libcanard #38: Suggestions II

olliw42, 2017-04-07:
I've just seen the Suggestions by OlliW comit :)

I would also suggest this change in canard.h

```
...
#include <stdbool.h>
#ifndef CANARD_ASSERT
#  include <assert.h>
#endif

#ifdef __cplusplus
...
```

On my system at least (baremetal, STM32F103, CocCoox) linking to the assert.h causes a chain of issues

In canard_stm32.c I found usleep() troubling, and did this change 

```
...
#include "canard_stm32.h"
#include "_internal_bxcan.h"
//#include <unistd.h>
extern void canardSTM32DelayUs(uint16_t);
...

...
        // Sleep 1 millisecond
//        usleep(1000);           // TODO: This function may be missing on some platforms
            canardSTM32DelayUs(1000);
    }
...
```

pavel-kirienko, 2017-04-07:
Thanks Olli!

Please observe that the first suggestion would break the build if `CANARD_ASSERT` is defined and the definition references `assert()`, so this solution would not work well. Instead I propose to investigate why inclusion of `assert.h` causes trouble with your build - it shouldn't, especially if `CANARD_ASSERT` is defined empty.

Concerning `usleep()`, I can see the value in your proposition; however, you can obtain the same result simply by redefining `usleep()` elsewhere in your application, e.g.:

```c
int usleep(useconds_t usec)
{
    // Place your busyloop here
}
```

olliw42, 2017-04-07:
CANARD_ASSERT: 
I agree it's not cool, and I also agree that one should investigate what the real problem is. However, I'm really not good at this. I've spend about 1h to figure out which lib the problem was, and that was all I was able to come up with. I guess I leave it at this. I thus change my statement from a suggestion to a note, that assert.h may cause problems on some systems :)

usleep(): ????
Just tried and put such a function in the demo.c, but got a "implicit declaration of usleep" warning, and I must admit not to my surprise ... so, on my end this doesn't seem to do the same thing ...

pavel-kirienko, 2017-04-07:
> usleep(): ????
> Just tried and put such a function in the demo.c, but got a "implicit declaration of usleep" warning, and I must admit not to my surprise ... so, on my end this doesn't seem to do the same thing ...

This is because you forgot to `#include <unistd.h>` in your `demo.c`.

olliw42, 2017-04-07:
indeed ... I can't say that I understand why, but indeed it's working, I was trying to avoid having to include unistd.h, since it is so dubious to me ... well ... I guess you see now that all this compiler-compatible-library-fancineses are way above my head ... and that a PR by me would cause more workload than it would reduce :)

I should better stick to the algorithms :)

Anyway, thx a lot. The libcanard v0.1 is a big step forward, IMHO.

closed 2017-04-07 by pavel-kirienko