OpenCyphal/libcanard #42: /usr/include/c++/5/limits:42:28: fatal error: bits/c++config.h: No such file or directory

yangang123, 2017-08-03:
[ 41%] Building CXX object CMakeFiles/run_tests.dir/stm32/test_can_timings.cpp.o
In file included from /usr/include/gtest/gtest.h:54:0,
                 from /home/yangang/work/cortex_m7_proj/Bootloader/libcanard/tests/stm32/test_can_timings.cpp:25:
**/usr/include/c++/5/limits:42:28: fatal error: bits/c++config.h: No such file or directory**
compilation terminated.
CMakeFiles/run_tests.dir/build.make:62: recipe for target 'CMakeFiles/run_tests.dir/stm32/test_can_timings.cpp.o' failed
make[2]: *** [CMakeFiles/run_tests.dir/stm32/test_can_timings.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/run_tests.dir/all' failed
make[1]: *** [CMakeFiles/run_tests.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

yangang123, 2017-08-03:
$ git submodule add https://github.com/UAVCAN/libcanard
$ mkdir build && cd build
$ cmake ../libcanard/tests
$ make

An Error Has Occurred !!!!!!!!!!!!!!!!!!!!

`/usr/include/c++/5/limits:42:28: fatal error: bits/c++config.h: No such file or directory`

pavel-kirienko, 2017-08-03:
In order to be able to compile 32-bit targets on a 64-bit system, you should install 32-bit libraries first. If you're on a Debian-based system, the missing packages are, probably, `libc6:i386` and `libstdc++6:i386` (perhaps some other are missing too).

Please observe that you don't have to build anything to use this library. Building unit tests (this is what you're doing) is useful only if you intend to contribute to the library.

closed 2017-08-03 by pavel-kirienko

yangang123, 2017-08-06:
pavel-kirienko:
      thank you very much!!!