OpenCyphal/libcanard #43: Tx Buffer in the driver?

mgiaco, 2017-09-05:
@pavel-kirienko Hi, this is more a generell architecture question.

I am wondering why you are added the TX and RX buffers inside the core of libcanard and not in the driver itself.  For a similar proprietary can implementation where I also must support different controllers  - I am not sure which way I should go.
  
Before I have looked into this implementation I taught i will use a TX and an RX Buffer inside the driver because I taught this way I could implement the send and receive with better performance and less memory consumption for the different controllers. 

The Infineon XMC as an example hast several Buffers which can be combined to queues for TX and RX. Also some NXP controllers have suche features as i know. And for the other targets I could implement the buffers inside the driver - so that was my plan.  

But now I am not sure if i should use a 2 stage buffering. ..

So why do you choose that design/architecture?

many thanks
mathias

pavel-kirienko, 2017-09-06:
Both Libuavcan and Libcanard share the same buffering architecture: the RX buffer resides in the driver (not in the core!), and the TX buffer resides in the core.

The motivation for keeping the RX buffer in the driver is quite simple: the buffer serves as a synchronization primitive between the hardware-related code (which is likely to be interrupt-driven) and the higher-level library code. We gain nothing by moving the buffer into the library, and lose in flexibility and simplicity.

The TX buffer is implemented in the core for several reasons:

1. Proper implementation of a TX buffer for CAN is harder than it seems at a first glance. The implementer has to ensure that the frames are prioritized correctly, that the buffer is not prone to the inner priority inversion problem, and that the buffer overflow errors are resolved properly. It is possible to implement all that in the driver; however, that would have made implementations of platform drivers a lot more difficult.
2. Additionally, having a TX buffer implemented in the core simplifies the interface between the core and the driver. Consider Libcanard: there are two simple functions designed for unloading the frames from the core into the driver: peek and pop. Peek returns a reference to the CAN frame that should be transmitted next; pop removes the topmost frame. If the buffer were located in the driver, the interface would have been much more complex.

closed 2017-09-06 by pavel-kirienko

mgiaco, 2017-09-07:
@pavel-kirienko many many thanks for this answer.

mgiaco, 2017-09-07:
Hi Pavel I have some more  questions:
1 So the frame prioritization only looked at the last entry on the queue do you think this is really a benefit?

Also i am wondering why you do not need any looking on peek push, pop  in relation with the frame prioritization? Let´s say you have to frames in the queue. Then there is some Tx ongoing and in the middle of the peek or so someone pushes a higher frame on the queue from a higher priority task and this happens maybe here https://github.com/UAVCAN/libcanard/blob/master/drivers/socketcan/socketcan.c#L97

Do you see no problem here?

pavel-kirienko, 2017-09-07:
The peek operation always returns the frame with the highest priority. The ordering according to the priority happens when a new frame is inserted into the queue.

There is no race condition because both libuavcan and libcanard are inherently single-threaded. If your application wants to use the library from multiple threads, you are responsible for correct synchronization.