OpenCyphal/libcanard #45: Potential Bug in canardSTM32ConfigureAcceptanceFilters() !?

olliw42, 2018-02-24:
I'm chasing an issue, which I believe might be due to a potential bug in the function canardSTM32ConfigureAcceptanceFilters().

It's in this code part:
https://github.com/UAVCAN/libcanard/blob/master/drivers/stm32/canard_stm32.c#L482-L493

The if looks incorrect to me.
For instance, if in both the id and the mask the EFF flags are not set, corresponding to STD frames, then nevertheless the first branch is executed, and not the second as it would be correct. This effectively prevents STD frames to be used with this filter setter function.

There are some comments in the code, but it is not totally clear that the code is supposed to do. My guess would be that the second branch should be called only when in both the id and the mask the EFF flag is not set, i.e. a STD frame is clearly called, in which case line 482 probably should be

`  if ((cfg->id & CANARD_CAN_FRAME_EFF) || (cfg->mask & CANARD_CAN_FRAME_EFF))`

pavel-kirienko, 2018-02-24:
Thanks, Ollie! The code is correct, but it is, indeed, quite counterintuitive. I just pushed a new commit that makes a better job at explaining what's going on. Please review it and reopen this ticket if you find any flaws in the provided explanation.

pavel-kirienko, 2018-02-24:
By the way, for the benefit of posterity, let me correct you on this:

> For instance, if in both the id and the mask the EFF flags are not set, corresponding to STD frames

This is false. (ID & EFF) == (Mask & EFF) == 0 means that both EFF and STD should be accepted. If you want only STD frames, you should set EFF in Mask, but not in ID.

closed 2018-02-24 by pavel-kirienko

olliw42, 2018-02-26:
thx
I see what you're saying
it's much easier now to understand the flow, and thus the usage. Before it appeared as if the if branch is just for handling this one special case, which couldn't be the whole truth
I think it would be helpful to add your last statement to the comments. Although it's clear after a while, it's a bit non-intuitive especially as the ids/masks are treated differently in the two cases. I mean, it's kind of odd that one has to set a EFF flag, but send both the id and mask in STDID format (<<21), while when the EFFs are not set, they both need to be send in EXTID format (<<3). It's correct as you say, but it's odd. Maybe that too could be briefly indicated.
cheers, Olli

commit 636795f4 (2018-02-24): STM32: better explanation of the STM32's bxCAN frame filtering logic; fixes #45