OpenCyphal/libcanard #46: Time-stamps for outgoing frames

pavelvasev, 2018-05-02:
According to http://uavcan.org/Specification/4._CAN_bus_transport_layer/#service-invocation
```
Service transfer transmission should be aborted if does not complete in 1 second.
```
As I understand, this means: if node determines it cannot send uavcan message within 1 second, it should stop trying for that particular message. If that true, then the following problem might have place.

Libcanard stores outgoing CAN frames in CanardTxQueueItem list which does not contain a time-stamp:
```
struct CanardTxQueueItem
{
    CanardTxQueueItem* next;
    CanardCANFrame frame;
};
```
Thus it seems impossible to determine whenever outgoing transfer item has reached a timeout.

### addition 1 
For multi-frame transfers it seems second and following frames may be stamped as first frame + 2 seconds, according to constant value in canard.c:
```
#define TRANSFER_TIMEOUT_USEC                       2000000
```
### addition 2
If timestamp will exist, then it will be possible to implement function like canardCleanupStaleTransfers but for outgoing transfers. So application may decide if it wants to clean up outgoing stale frames of keep them going.

pavel-kirienko, 2018-05-07:
That feature has been removed intentionally for reasons of simplicity. Simple applications can get by with global transmission timeout management for the entire queue rather than per-frame since the specification does not make timeout management a hard requirement (it says "should" not "must"). If this is not suitable for your application, consider using libuavcan.

closed 2018-05-07 by pavel-kirienko

pavelvasev, 2018-05-07:
Pavel, could you please explain the approach of `global transmission timeout management for the entire queue`? I just dont understand what it means..

pavel-kirienko, 2018-05-07:
*Global transmission timeout management for the entire queue* can be implemented as follows: If no frames could be emitted in 1 seconds, flush the queue.

pavelvasev, 2018-05-07:
Thank you, Pavel!