OpenCyphal/libcanard #50: Missing support for DSDL generator for libcanard

storola, 2018-05-22:
Missing support for DSDL generator for libcanard

This would be useful because:
- Easier way to use UAVCAN with libcanard
- UAVCAN messages packed and un-packed systematic way
- Allows UAVCAN usage with MCU's having smaller flash & memory size
- ANSI C code library
- Makes possible to write cross test generator between uavcan & libcanard pack and unpack functions

PR: coming in soon in near future

pavel-kirienko, 2018-05-22:
Yes, DSDL code generation is sorely missing. There is an abandoned effort initiated by @antoinealb years ago, perhaps you could use it as a reference (or not, I'm not sure): https://github.com/UAVCAN/libcanard/pull/6

Minor nitpick:

> ANSI C code library

Libcanard is not ANSI C, it's written in C99.

antoinealb, 2018-05-22:
Yeah, I dont think my aborted attempt would be useful to someone willing to start now honestly.

kjetilkjeka, 2018-05-22:
I was playing with the thought of adding a c target to the [Rust DSDL compiler](https://crates.io/crates/dsdl_compiler).

Right now I haven't had the chance to work on this project for many months but in the case that someone would like to do it, I would be able to take the time to mentor such task.

storola, 2018-05-22:
I have a solution ready. It is basically a fork from the uavcan_dsdl generator. I need to do some cleanup rounds and I am ready to push it for the review.

There are few things which I don't like, but I do have not really good solution around that. Missing namespace in C, which means that I it is generating long function and define names, but I can live with that, it just looks a bit ugly.

storola, 2018-05-30:
some bureaucracy delays, eta 1 week

kjetilkjeka, 2018-06-19:
@storola do you have a status update?

storola, 2018-06-19:
still reason for the delay, but should be ready to on next week. Wish me luck :)

storola, 2018-06-29:
Finally proposal is out!
Here you can find some info about the usage.
https://github.com/storola/libcanard/tree/dsdl_compiler/dsdl_compiler

pavel-kirienko, 2018-10-04:
Fixed in #53

closed 2018-10-04 by pavel-kirienko