OpenCyphal/libcanard #51: Wrong size in CANARD_STATIC_ASSERT(sizeof(CanardRxState) <= 28)?

storola, 2018-05-29:
https://github.com/UAVCAN/libcanard/blob/7899ddf80a5981363e1f6e72ef7df2ae9359ac0f/canard.h#L238

CANARD_STATIC_ASSERT(sizeof(CanardRxState) <= 28, "Invalid memory layout");

We have a few environments and CanardRxState size seems to be aligned to be 32
Meaning assert will fail. When c11 used. Even when compiling with c99 CanardRxState size is reported as 32 bytes.

Should this be sizeof(CanardRxState) <= 32 ?

storola, 2018-05-29:
STM32
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 6.2.1 20161205 (release) [ARM/embedded-6-branch revision 243739]

pavel-kirienko, 2018-05-29:
I suggest replacing `sizeof(CanardRxState) <= 28` with `offsetof(CanardRxState, buffer_head) <= 28`.

pavel-kirienko, 2018-05-29:
Could you submit a PR please?

storola, 2018-05-29:
Here we go
https://github.com/UAVCAN/libcanard/pull/52

closed 2018-05-29 by pavel-kirienko