OpenCyphal/libcanard #54: Teensy Drivers instead of AVR

mjs513, 2018-08-02:
in the Libcanuav GitHub library there is a driver for the Teensy boards (NXP).  Can that be used in conjunction with this library?

pavel-kirienko, 2018-08-02:
If you're referring to libuavcan, it doesn't support that hardware platform yet (although there is a work in progress). Even if it did, a driver designed for libuavcan would probably be unnecessarily complex for the needs of libcanard. I suggest you write your own send/receive routines, it should be really straightforward; use this as a reference: https://github.com/UAVCAN/libcanard/tree/master/drivers/stm32

Hope this helps!

closed 2018-08-02 by pavel-kirienko

mjs513, 2018-08-02:
@pavel-kirienko   Really appreciate getting back to me on this.  Actually what I would like to do is get this, https://github.com/tum-phoenix/drive_teensy_libuavcan, working on Teensy with the use of cmake.  What I am sure about and is confusion, is what would an Arduino style library look like for libuavcan as opposed to using cmake.  

I wanted to start with libcanard since is seemed more appropriate as an initial test with the ultimate goal of getting libuavcan working.  Any help or pointing me in the right direction would help.  I have already sent ahrs and GPS data over can successfully but what to give an official version up and running.

Thanks
Mike

mjs513, 2018-08-02:
Actually I am working with Teensy 3.5, not an Arduino.  The Teensy actually uses a GCC C++11 toolchain.  I am really just using the Arduino IDE to compile the program and load it on the Teensy.  For the Teensy there is an addin called Teensyduino that makes all the changes.  Well anyway, you got me sidetracked.

If I can stick with the Arduino IDE/Teensyduino it will may get more usage in the Teensy community.  I have done it with Mavlink tied to APM planner and I know there are folks using the Teensy for flight controllers.  Again I am rambling sorry.

> if you followed the cmake script you'd see that the build process (after you remove all of the unit-testing-related shenanigans) is quite straightforward:

I will take your advice and dig into the cmake's

Thanks for your help 
Mike