OpenCyphal/libcanard #55: Support for MCUs, where smallest addressable memory cell  is 16-bits.

HDKLog, 2018-09-17:
On some MCUs like TMS320F2812  smallest addressable memory cell  is 16-bits and there is no definition of uint8_t or int8_t, which cause compilation failure.

pavel-kirienko, 2018-09-17:
Woah! Are there people in the world who care about TMS320 enough to submit a PR fixing this?

The fix won't be trivial I'm afraid. Data structures employed by UAVCAN frequently require 1-byte granularity, so a lot of logic will have to be added to work around that. Perhaps you're better off using a less exotic architecture for CAN communication?

thirtytwobits, 2018-09-18:
Hmm, I'm not sure if the Piccolo series has this constraint. That'd be the only relevant product I could think of that would be relevant to this project.

pavel-kirienko, 2018-09-18:
Please re-open if still relevant.

closed 2018-09-18 by pavel-kirienko