OpenCyphal/libcanard #59: Unnamed structs gives error when compiling with -std=c99

aasmune, 2018-10-18:
When compiling using `-std=c99` I get the following error related to how the structs for dynamic arrays are generated in the header files:
```Bash
$ cmake . && make
running cd "/home/aasmune/Documents/git/UAVCAN-examples" && /usr/bin/python3 lib/libcanard/dsdl_compiler/libcanard_dsdlc --outdir dsdl lib/dsdl/uavcan/  2>&1
-- Configuring done
-- Generating done
-- Build files have been written to: /home/aasmune/Documents/git/UAVCAN-examples
Scanning dependencies of target example
[  1%] Building C object CMakeFiles/example.dir/src/example.c.o
In file included from /home/aasmune/Documents/git/UAVCAN-examples/dsdl/uavcan/protocol/GetNodeInfo.h:20:0,
                 from /home/aasmune/Documents/git/UAVCAN-examples/src/example.c:23:
/home/aasmune/Documents/git/UAVCAN-examples/dsdl/uavcan/protocol/HardwareVersion.h:71:6: error: ISO C99 doesn’t support unnamed structs/unions [-Werror=pedantic]
     };
      ^
In file included from /home/aasmune/Documents/git/UAVCAN-examples/src/example.c:23:0:
/home/aasmune/Documents/git/UAVCAN-examples/dsdl/uavcan/protocol/GetNodeInfo.h:99:6: error: ISO C99 doesn’t support unnamed structs/unions [-Werror=pedantic]
     };
      ^
/home/aasmune/Documents/git/UAVCAN-examples/src/example.c: In function ‘shouldAcceptTransfer’:
/home/aasmune/Documents/git/UAVCAN-examples/src/example.c:202:56: warning: unused parameter ‘ins’ [-Wunused-parameter]
 static bool shouldAcceptTransfer(const CanardInstance* ins,
                                                        ^~~
cc1: all warnings being treated as errors
CMakeFiles/example.dir/build.make:62: recipe for target 'CMakeFiles/example.dir/src/example.c.o' failed
make[2]: *** [CMakeFiles/example.dir/src/example.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/example.dir/all' failed
make[1]: *** [CMakeFiles/example.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

The error is due to how dynamic arrays are created:
```C
// GetNodeInfo.h
...

typedef struct
{
    // FieldTypes
    uavcan_protocol_NodeStatus status;                        //
    uavcan_protocol_SoftwareVersion software_version;              //
    uavcan_protocol_HardwareVersion hardware_version;              //
    struct
    {
        uint8_t    name_len;                  // Dynamic array length
        uint8_t*   name;                      // Dynamic Array 8bit[80] max items
    };

} uavcan_protocol_GetNodeInfoResponse;
...
```
It will compile using `-std=c11`, but I think it should be fixed as Libcanard uses the Zubax C++ coding conventions which states that C code should be written in standard c99.

antoinealb, 2018-10-18:
I think this is only the case in pedantic mode, at least for GCC [1]. Can you confirm that you are building using `--pedantic` ? And should we test that in Travis ?

[1] http://openbsd-archive.7691.n7.nabble.com/allow-anonymous-structs-unions-with-gcc-std-c99-td264402.html

aasmune, 2018-10-18:
I am using the same build parameters as is used to build the demo.c application, so pedantic is indeed used.

pavel-kirienko, 2018-10-18:
Indeed, the way we generate code should be changed. I propose either of the following options.

```c
typedef struct
{
    uavcan_protocol_NodeStatus status;
    uavcan_protocol_SoftwareVersion software_version;
    uavcan_protocol_HardwareVersion hardware_version;
    struct
    {
        uint8_t    len;
        uint8_t*   data;
    } name;
} uavcan_protocol_GetNodeInfoResponse;
```
```c
typedef struct
{
    uavcan_protocol_NodeStatus status;
    uavcan_protocol_SoftwareVersion software_version;
    uavcan_protocol_HardwareVersion hardware_version;
    uint8_t    name_len;
    uint8_t*   name;
} uavcan_protocol_GetNodeInfoResponse;
```

What we have now is also a minor deviation from the objectives stated in the [design document](https://github.com/UAVCAN/libcanard/blob/master/DESIGN.md#code-generation).

antoinealb, 2018-10-18:
I think only your second proposal is compatible with the linked design document, isn't it ? Also `uint8_t` is not enough for the `_len` field, I think it should be `size_t` or `uint32_t` at least.

pavel-kirienko, 2018-10-18:
> I think only your second proposal is compatible with the linked design document, isn't it ?

Yeah. But then again, design documents are made to be changed.

> Also uint8_t is not enough for the _len field, I think it should be size_t or uint32_t at least.

The compiler chooses between `uint8_t` and `uint16_t` depending on the maximum size. No sensible UAVCAN application is expected to ever use more than 65535 items in an array (consider the best case, an array of bits: that would be 65535/8 = 8 KB of data), so I'm not sure whether `size_t` is actually a good choice, but I might be wrong.

pavel-kirienko, 2018-10-18:
On the other hand, there are safety critical coding guidelines which occasionally state that data types should never be wider than necessary to reduce the amount of invalid states in the system (I just spent 15 minutes looking for the source but couldn't find it; neither MISRA C++ nor JPL seem to require that, perhaps I read that in one of the ESA documents, can't recall). I suggest keeping it as is right now and re-open this conversation in a separate ticket if desired.

aasmune, 2018-10-18:
Personally, I like your first proposal best. I think it is better to gather both the data and the length of a dynamic array into a struct to make the outer struct clean.

>I would prefer that it does not change depending on the maximum size. Otherwise, some working code could silently start breaking when changing the max length of the array.

As of my understanding, to change the max length of an array requires that the DSDL definition need to be updated. However, by changing the max length, these two DSDL versions no longer are compatible with each other. Hence in a case where the max size of an array has been changed the project need to start using the new DSDL version and should make sure that the update doesn't break the existing behavior.

pavel-kirienko, 2018-10-18:
> However, by changing the max length, these two DSDL versions no longer are compatible with each other.

We have a [highly relevant discussion on the forum](https://forum.uavcan.org/t/the-case-for-code-compatibility/194) that you might want to have a peek at.

pavel-kirienko, 2018-10-18:
> I like your first proposal best. I think it is better to gather both the data and the length of a dynamic array into a struct to make the outer struct clean.

I agree. If @antoinealb doesn't object then let's go with that.

aasmune, 2018-10-18:
> We have a highly relevant discussion on the forum that you might want to have a peek at.

Yes, I am aware of that discussion.

kjetilkjeka, 2018-10-18:
> On the other hand, there are safety critical coding guidelines which occasionally state that data types should never be wider than necessary to reduce the amount of invalid states in the system (I [...]

I can't express how deeply I share this sentiment. Perhaps it's not surprising that the Rust guy would like to statically guarantee the invariants through the type system. At the same time, I'm honestly not sure how much value this has in a weakly (perhaps implcit type conversions express more clearly what I mean) typed language like C. If you attempt to write 1080 to your `uint8_t` your invariants will most likely be invalidated and I suspect you might be worse off than if you were using an `uint16_t` in the first place. I guess to preserve the invariant you will need some runtime checking (maybe unless you're using a language with dependent types) and handling, which again is quite manual and laboursome in C.

> I would prefer that it does not change depending on the maximum size. Otherwise some working code could silently start breaking when changing the max length of the array. For size_t vs uint16_t I don't really have any argument here. Just observing that size_t is uint32_t on ARMs (where it would probably be padded to 32 bits anyway), and uint16_t on AVRs (so the size that you would pick if you don't want the type to be automatically changed by the dsdlc). Hence I would go with size_t.

I fully agree, `size_t` will probably also convey intetion a bit better so I would have a small preference against using it.

antoinealb, 2018-10-18:
> > I like your first proposal best. I think it is better to gather both the data and the length of a dynamic array into a struct to make the outer struct clean.
> 
> I agree. If @antoinealb doesn't object then let's go with that.

Ok for me.

closed 2018-10-20 by pavel-kirienko