OpenCyphal/libcanard #61: Make Travis (or another CI platform) test the DSDL compiler as well

pavel-kirienko, 2018-10-18:
We should add a fixture that runs the DSDL compiler against some data types (e.g. the standard set) and then uses them in a libcanard application.

aasmune, 2018-10-20:
In addition to system tests of the DSDL compiler, unit tests for all generated C functions, for all the data types in the uavcan namespace, should also be written. This will allow validating the behavior of each generated encode/decode function for all data types. Unit tests should not require any physical HW, as it only compares the output of each function a precalculated correct output. This will make it quite simple to set up Travis to run these tests on-commit. 

It will require some work to write enough unit tests to get an acceptable coverage percentage of all the data types, but it should be considered to minimize the possibilities that PRs don't break compatibility, or contains bugs.

In my opinion, it should also be written unit tests for the rest of the functions in Libcanard, now as testing is becoming more important for v1.0.

pavel-kirienko, 2019-09-12:
Closed because #137

closed 2019-09-12 by pavel-kirienko