OpenCyphal/libcanard #62: Inconsistency between design documentation and generated C definition for dynamic arrays

aasmune, 2018-10-18:
Reading through the [design documentation](https://github.com/UAVCAN/libcanard/blob/master/DESIGN.md#code-generation)  of Libcanard, I see that the corresponding C definition of a dynamic array should be an array with the maximum number of elements which is allowed in the dynamic array. E.g `Foo[<5] bars` in DSDL should become `Foo bars[4];` in C. 

As of now, dynamic arrays are created as pointers, not as arrays. Is this the desired behavior, or should the pointers be replaced with the definition stated in the design documentation?

pavel-kirienko, 2018-10-18:
I don't have any objections against this deviation. As to the motivation, we should ask @storola.

kjetilkjeka, 2018-10-18:
I asume the reason is to allow application designers to use a 32 byte buffer even if the datatype supports 256. 

I agree it's a bit weird that the behavior is different between composite types and arrays and even more between dynamic arrays and static arrays. And I would prefer that this was made more consistent.

pavel-kirienko, 2018-10-20:
This issue is somewhat hanging in an odd position where it's neither a bug nor a discussion. Do we want to change the behavior to what is specified in the design document?

Pros: simpler and less error-prone API; cons: higher memory footprint.

On one hand, Libcanard tends to attract people who are stuck with their memory-impaired microcontrollers, so that change might be seen as a regression. On the other hand, if one can't afford an extra couple of hundred bytes on the stack, they are probably better off serializing stuff manually than asking a computer to do that for them.

kjetilkjeka, 2018-10-20:
> On the other hand, if one can't afford an extra couple of hundred bytes on the stack, they are probably better off serializing stuff manually than asking a computer to do that for them.

I agree with this point of view. The "sensible default" should definitely lean towards ease of use.

Perhaps we can give the 8bit guys some extra control by allowing a preprocessor define called `DSDLC_DYNAMIC_ARRAYS_AS_POINTERS` or similar to opt into the manual version?

pavel-kirienko, 2018-10-20:
> Perhaps we can give the 8bit guys some extra control by allowing a preprocessor define called `DSDLC_DYNAMIC_ARRAYS_AS_POINTERS` or similar to opt into the manual version?

Uh oh, looking at the API I can't imagine how that can be done easily. The external API of serialization functions depends on that quite a bit.

storola, 2018-10-22:
> As of now, dynamic arrays are created as pointers, not as arrays. Is this the desired behavior, or should the pointers be replaced with the definition stated in the design documentation

This is desired behavior. Dynamic arrays in C are pointers, I don't see any problem with this approach. I wanted to create library work like this, this allows users to allocate memory as little as needed. This allows a lot of control for the user, it is not strict to any heap allocation like malloc / alloc (but can be used if used in system). Using this approach it also unfortunately gives responsibility to user to reserve enough space for the dynamic arrays.

> I agree it's a bit weird that the behavior is different between composite types and arrays and even more between dynamic arrays and static arrays. And I would prefer that this was made more consistent.

In static vs dynamic array, it works a differently because there is no possibility to save RAM in static array definition, but you need to reserve that always, even if not needed.

So overall I don't want to this generated approach to be changed as it gives a lot of flexibility. True, it is not easiest API to use, but you have way bigger libuavcan for that. So if wanted maybe documentation could be updated to make this more clear.

pavel-kirienko, 2018-12-10:
Nobody seems to mind, closing.

closed 2018-12-10 by pavel-kirienko