OpenCyphal/libcanard #64: DSDL compiler: incorrect array serialization

pavel-kirienko, 2018-10-20:
A [bug in the updated DSDL compiler has been found](https://forum.uavcan.org/t/there-is-a-bug-in-libcanard-dsdl-please-check/205):

> uavcan/equipment/actuator/actuator_ArrayCommand.c: In function ‘uavcan_equipment_actuator_ArrayCommand_encode_internal’:
> 
> uavcan/equipment/actuator/actuator_ArrayCommand.c:41:93: error: subscripted value is neither array nor pointer nor vector
> offset += uavcan_equipment_actuator_Command_encode_internal((void*)&amp;source-&gt;commands[c], msg_buf, offset, 0);
> 
> commands[c] is wrong ,that is commands.data[c] is right.

@aasmune could you look into this please?

aasmune, 2018-10-20:
It looks like I forgot to update some parts of the template, and I didn't notice as I did not use the affected messages. I have now fixed the reported bug (and two more). I looked through the compiled DSDL code after the fixes, and it looks good to me now. However it could be that I have overlooked something, so please check again just to be sure.

pavel-kirienko, 2018-10-20:
Thanks @aasmune, could you send a pull request please?

closed 2018-10-20 by pavel-kirienko