OpenCyphal/libcanard #66: I found a bug in libcanard dsdl compiler

Cherish-Gww, 2018-10-30:
Hi
Today I use libcanard dsdl compiler to generate code,
I fill below in my *.uavcan files:

uint8[<=6] command

and I found that in encode,
uavcan_xxx_encode_internal(source, msg_buf, offset, 1);
uavcan_xxx_encode_internal(uavcan_syrius_RobotCommand* source, void* msg_buf, uint32_t offset, uint8_t root_item)

Here is root_item = 1
and here inside uavcan_xxx_encode_internal
if (! root_item)
{
// - Add array length
canardEncodeScalar(msg_buf, offset, 3, (void*)&source->command.len);
offset += 3;
}

I think if (! root_item) need to be if (root_item).
this means that the first 3bits means length of my buffer.

because that in decode, this parameter need it.

pavel-kirienko, 2018-10-30:
@storola could you have a look perhaps? Thanks!

pavel-kirienko, 2018-11-27:
@Cherish-Gww  Please reopen if relevant.

closed 2018-11-27 by pavel-kirienko