OpenCyphal/libcanard #83: There is a Bug in libcanard dsdl,Please check –BeginFirmwareUpdateRequest

Cherish-Gww, 2019-02-15:
There‘s a Bug in uavcan_protocol_file_BeginFirmwareUpdateRequest_decode_internal(…)
This Bug lead to a wrong decode struct.
line 89
// Compound
offset = uavcan_protocol_file_Path_decode_internal(transfer, 0, &dest->image_file_remote_path, dyn_arr_buf, offset);

The second actual parameter should not be ‘0’, It should be ‘payload_len’;

fix:
// Compound
offset = uavcan_protocol_file_Path_decode_internal(transfer, payload_len, &dest->image_file_remote_path, dyn_arr_buf, offset);

pavel-kirienko, 2019-02-17:
@aasmune how does it look to you?

aasmune, 2019-02-17:
After looking briefly at this, it looks to me that this bug affects decoding of all types (both messages and services) which contain at least one compound type. If this is the case, then the zero in this line should be changed to 'payload_len' https://github.com/UAVCAN/libcanard/blob/master/dsdl_compiler/libcanard_dsdl_compiler/code_type_template.tmpl#L320. Are you able to verify that this is the case @Cherish-Gww?

PR #86 (2019-02-21): Fix bug with decoding of compound structs
This PR fixes #83.
closed 2019-02-21 by pavel-kirienko