OpenCyphal/libcanard #95: Optional Tx queue

mike7c2, 2019-03-18:
The tx queue is a bit of a pain point for carrying out integration - as there is no "push notification" to the queue it must be polled.

I have a branch which makes the TX queue optional via a build flag, `CANARD_TX_NO_QUEUE`, I wonder if a change like this would be acceptable to the project?

https://github.com/mike7c2/libcanard/tree/optional_tx_queue

I looked at other ways to resolve this and I think adding a "queue no longer empty" style notification could solve the issue - in our specific application though ignoring priority and pushing directly to an RTOS queue is the way we want to go.

pavel-kirienko, 2019-03-19:
I've looked at your changes and I think they add too much variance to the build process, making the code harder to maintain and test. Also, the design where you push outgoing frames into a FIFO queue is wrong and we should not endorse it through the API. One who doesn't want to poll the TX queue can always implement a push notification outside of the library, e.g. by invoking some custom handler after they invoke `canardBroadcast()` or `canardRequestOrRespond()` (there is no other way for new TX frames to appear in the queue).

I am mildly inclined to reject this proposal. @thirtytwobits @kjetilkjeka @aasmune what do you think?

spiremike, 2019-03-19:
I share your feelings about adding variance to the build process, and happy to look at other options here.

The biggest issue with the current setup is that we must always have enough space to buffer ALL of the frames for any message emitted from the library - this isn't a big deal normally but becomes annoying when trying to send larger messages.

Having a blocking queue means that a thread of execution can be generating frames as another thread (whether through RTOS or interrupts) is sending them - this allows messages of arbitrary size to be sent even with a buffer size of 1 frame. 

This also aids determinism of receiving messages because the TX and RX queues both rely on the same frame pool - currently it's possible to have a system which will behave well almost all the time, until it emits a long message at which point reception of other longer messages can fail.

pavel-kirienko, 2019-03-19:
I see what you are saying. Should we explore the possibility of adding a TX push notification callback to the library alongside `should_accept` and `on_reception`, keeping everything else unchanged?

spiremike, 2019-03-19:
Let me have a bit more of a dig and ruminate on this - we would need to make the queue thread safe as well and there will be some trickiness making it behave in a blocking manner too (might need hooks to allow the user to semaphore count messages in and out of the queue and critical sections to make the queue itself thread safe).

I'm fully open to any suggestions which achieve the end goal!

kjetilkjeka, 2019-03-19:
I haven't been able to look at the suggestion in detail, but I also got the impression that the original suggestion did not fit well into libcanard. I also agree that adding TX push notification is a nice addition, and would prefer going in this direction.

pavel-kirienko, 2020-03-08:
@mike7c2 is there interest in having this implemented in Libcanard v1?

pavel-kirienko, 2020-10-12:
Closing as stale.

closed 2020-10-12 by pavel-kirienko