OpenCyphal/libcanard #98: Rename API entities to enhance compliance with the spec

pavel-kirienko, 2019-03-24:
* `canardEncodeScalar` --> `canardEncodePrimitive`
* `canardDecodeScalar` --> `canardDecodePrimitive`
* `CanardTransferTypeResponse` --> `CanardTransferTypeServiceResponse`
* `CanardTransferTypeRequest` --> `CanardTransferTypeServiceRequest`
* `CanardTransferTypeBroadcast` --> `CanardTransferTypeMessagePublication`
* `canardBroadcast` --> `canardPublishMessage`

pavel-kirienko, 2019-03-27:
From #99:

> The function "canardCleanupStaleTransfers()" must be removed (this affects the API and the demo); see section 4.2.1 for details. This change to the spec was implemented in order to improve deterministic behavior at the cost of somewhat increased average-case memory usage (the worst case is unaffected). The constant `CANARD_RECOMMENDED_STALE_TRANSFER_CLEANUP_INTERVAL_USEC` is therefore not needed either (there could be other constants which we no longer need).
> 
> Data type signature must be removed from everywhere. This includes the callback signature `CanardShouldAcceptTransfer`.
> 
> The API functions still use "data type ID" instead of "subject ID" and "service ID", this should be fixed.
> 
> The transfer priority constants require updating. We have proper mnemonic names for the eight priority levels defined in 4.1.4.
> 
> The script `show_data_type_info.py` will no longer be needed, since there are no signatures to worry about.

aasmune, 2019-03-28:
I have started working on this now but will wait to commit/push until #99 is merged.

aasmune, 2019-03-31:
Fixed in #106

closed 2019-03-31 by aasmune