OpenCyphal/libcanard #108: Remove TAO

aasmune, 2019-03-28:
The TAO (tail array optimization) logic needs to be removed (finally!). Would you like to do it in a new PR?

_Originally posted by @pavel-kirienko in https://github.com/UAVCAN/libcanard/pull/106_

aasmune, 2019-04-02:
What need to be done in this issue @pavel-kirienko ?

pavel-kirienko, 2019-04-02:
The DSDL compiler generates a bunch of tricky checks and branches that determine whether one of the variable-length arrays contained in the data type object should be encoded without being prefixed by its length field. All this needs to be removed, and every array should be encoded uniformly with its length field added always.

aasmune, 2019-04-02:
So if I have understood it correctly: All dynamic arrays should be encoded with first the number of items, then all the values?

pavel-kirienko, 2019-04-02:
Correct. Relevant context:

1. The old mailing list thread: https://groups.google.com/forum/#!topic/uavcan/guUjFsHk2Lo; search for the following text:
> I think this is the perfect occasion to share a revelation I had some time ago, a few years after announcing the first draft of the standard:
> TAO was a mistake.

2. https://github.com/UAVCAN/uavcan.github.io/issues/34

aasmune, 2019-04-02:
Fixed in the current DSDL compiler. A proper solution will be to change to using pydsdlgen. Leaving the issue open awaiting the proper fix.

closed 2020-01-17 by pavel-kirienko