OpenCyphal/libcanard #111: Missing constants in generated C files from DSDL

aasmune, 2019-04-01:
I have noticed that the generated C file doesn't contain any constants if the source DSDL file consists only of constants, without any other fields.

E.g. the following file will be created as [Foo.1.0_no_constants.txt](https://github.com/UAVCAN/libcanard/files/3029658/Foo.1.0_no_constants.txt) (converted from .h to .txt to be able to upload to Github and renamed)
```
uint8 BAR = 123
```

However, the following file will be created as [Foo.1.0_with_constant.txt](https://github.com/UAVCAN/libcanard/files/3029673/Foo.1.0_with_constant.txt)
```
uint8 BAR = 123
uint8 placeholder
```
I haven't looked deeply on the output from pydsdl during parsing, but is this expected behavior, or a bug? It should also be mentioned that this is with an "old" version of pydsdl (before v0.4)
@pavel-kirienko

pavel-kirienko, 2019-04-01:
> is this expected behavior, or a bug

It's a bug :beetle:

pavel-kirienko, 2019-04-01:
This question reminds me that maybe the code generator should be based on https://github.com/UAVCAN/pydsdlgen

aasmune, 2019-04-01:
> > is this expected behavior, or a bug
> 
> It's a bug 🐞

Do you know if its a bug with pydsdl or with the DSDL compiler in Libcanard?

> This question reminds me that maybe the code generator should be based on https://github.com/UAVCAN/pydsdlgen

I have also considered that we might should change to using that when it's ready.

pavel-kirienko, 2019-04-01:
> I have also considered that we might should change to using that when it's ready.

Should we ask @thirtytwobits if it's ready yet?

aasmune, 2019-04-02:
As I guess it will be some time before we change to using pydsdlgen I opened a PR, #113, with a fix for the current DSDL compiler.

aasmune, 2019-04-02:
Fixed in #113

closed 2019-04-02 by aasmune