OpenCyphal/libcanard #117: Compilation on 64 bit platforms fails

windelbouwman, 2019-04-25:
I tried to build an application with libcanard with a 64 bits compiler. Doing so will trigger static asserts, which is good, but is there any reason why we cannot use it on 64 bit? And, would it be possible to modify libcanard such that is can be compiled on 64 and 32 bit platforms?

kjetilkjeka, 2019-04-27:
The reason why it currently does not work is a static block size that expect a specific (maximum) pointer size. 

If/When we change to run-time (init) sized blocks, 64-bit arch can be supported somewhat easily by sizing the blocks depending on pointer size. This is currently not prioritized, but I know of other applications where this is beneficial as well, and I expect it to be implemented eventually.

windelbouwman, 2019-04-27:
Okay, if you could point me to the proper sourcecode to modify, I will create a pull request for this. What I had to do now was compile my application on ubuntu 64 bit as a 32 bit application. This required me to install some multi lib libraries. Also, since I was running in WSL (windows subsystem for linux) on windows, I had to use qemu-i386 to run my application. This were some hoops I had to jump, if we can make libcanard 32 and 64 bit compatible, this would be easier..

kjetilkjeka, 2019-04-28:
I will do it once possible. This is currently blocked on switching to dynamically sized blocks, before doing this switch It's a big advantage to update the framing. Updating the framing to the new protocol is what I'm attempting to get done today.

pavel-kirienko, 2019-04-28:
As Kjetil has already explained, [64-bit platforms are not supported by design](https://github.com/UAVCAN/libcanard/blob/master/DESIGN.md) in order to reduce memory footprint on other platforms (where the pointer size is smaller). This is important for libcanard since it is designed specifically for small embedded systems. Are you sure you really need libcanard on your GNU/Linux machine? Libuavcan is more feature-complete and is much easier to use than libcanard, have you considered using that instead?

windelbouwman, 2019-04-28:
You are right that we should pick libuavcan on GNU/linux machines. This would be our choice when we had a GNU/linux/windows application. However, the use case that we face is that we actually have a microcontroller application and want to run the same codebase (including the libcanard uavcan stack) on GNU/linux in order to test the application on a host only system. This gives several benefits such as better debugging tools, no cabling and hardware issues to name a few.

debosvi, 2019-05-16:
shall we disable, waiting new updates/pull requests, static_assert with custom config file?
in others words, it won't be an issue with structs alignment?

pavel-kirienko, 2019-05-16:
> in others words, it won't be an issue with structs alignment?

Don't do that, things will break horribly. Context: https://github.com/UAVCAN/libcanard/blob/master/DESIGN.md

pavel-kirienko, 2020-03-08:
Fixed in #142

closed 2020-03-08 by pavel-kirienko