OpenCyphal/libcanard #118: Include paths in generated sources contain backslashes

windelbouwman, 2019-04-25:
The source files which are generated by the dsdl compiler contain include paths with backslashes. This is a platform specific thing, but in general, we can always use forward slashes for include paths.

Problem is the function `type_output_filename` which uses forward slashes or backslashes depending on os. This is a problem if the generated files are generated in windows, then checked in into git, then compiled on linux. linux only supports forward slashes as path seperator. Windows supports both forward and backward slashes.

kjetilkjeka, 2019-04-27:
I expect not checking in the compilation artifacts from the DSDL compiler will be the default, and compiling for the host OS is the "sensible default". Perhaps the best solution would be to pass the type of separator as a flag to the application. Then if none is provided, the application can default to the OS specific.

Is this an acceptable solution @windelbouwman ?

windelbouwman, 2019-04-27:
This might be an acceptable solution. Even though, I think that the include directives always should contain only forward slashes, and never backslashes, therefore we can make this the default, and have no option for it.

See also:
https://stackoverflow.com/questions/5790161/is-the-backslash-acceptable-in-c-and-c-include-directives

aasmune, 2019-04-27:
> This might be an acceptable solution. Even though, I think that the include directives always should contain only forward slashes, and never backslashes, therefore we can make this the default, and have no option for it.

I agree, forward slashes should be used as default for all OS. I don't really see the benefits of having the option to change between them.

pavel-kirienko, 2019-04-28:
I agree with Asmund here. Paging @thirtytwobits because it's probably relevant for pydsdlgen.

closed 2019-09-12 by pavel-kirienko