OpenCyphal/libcanard #121: DSDL compiler error UAVCAN-v1.0

fvantienen, 2019-05-09:
When trying to compile the latest `public_regulated_data_types` using the `dsdl_compiler` in the uavcan-v1.0 branch (for both the data_types and libcanard) it will fail building with the following error:
```
pydsdl.regular_grammar_matcher.InvalidGrammarError: Invalid grammar: IOStatistics.0.1[<=MAX_PHYSICAL_INTERFACES] physical_interface_statistics
```

pavel-kirienko, 2019-05-09:
This is because it needs a newer version of PyDSDL.

fvantienen, 2019-05-09:
Any advise on how to fix this correctly or temporarily?
I already tried several combinations of different DSDL compiler and branches without any success. Not sure how the structure of this whole UAVCAN project and version numbering etc. is.

pavel-kirienko, 2019-05-09:
Just run `pip install pydsdl --upgrade` and you should be all set.

fvantienen, 2019-05-09:
I guess it is not that simple:
```
pydsdl/_expression.py", line 564, in <module>
    typing.Callable[[BinaryOperator[OperatorOutput]], BinaryOperator[OperatorOutput]]:
  File "/usr/lib/python3.5/typing.py", line 815, in __getitem__
    raise TypeError("This Callable type is already parameterized.")

```

pavel-kirienko, 2019-05-09:
You should get a newer Python. I first encountered that issue yesterday: https://github.com/UAVCAN/pydsdl/pull/26

fvantienen, 2019-05-09:
Ok 1 step forward but still having problems:
```
>>> from pydsdl import parse_namespace
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'parse_namespace'
```

pavel-kirienko, 2019-05-09:
Ah, that's because there have been API changes in the recent versions. They are outlined here: https://github.com/UAVCAN/pydsdl/pull/18/files?file-filters%5B%5D=.cfg&file-filters%5B%5D=.md&file-filters%5B%5D=.parsimonious&file-filters%5B%5D=.py&file-filters%5B%5D=.sh&file-filters%5B%5D=No+extension#diff-04c6e90faac2675aa89e2176d2eec7d8

fvantienen, 2019-05-09:
Ok I tried, but these are extreme changes and a lot has to be changed. So I give up for now, but is there any other way?

pavel-kirienko, 2019-05-09:
Åsmund said yesterday that he is planning to continue his work on libcanard for UAVCAN v1.0 soon: https://github.com/UAVCAN/libcanard/pull/119#issuecomment-490641692

[image]

closed 2019-09-12 by pavel-kirienko