OpenCyphal/libcanard #123: DSDL libcanard-compatible C source files or headers

williamyoung0, 2019-05-28:
`uint32_t uavcan_equipment_gnss_Fix_encode_internal(uavcan_equipment_gnss_Fix* source,
  void* msg_buf,
  uint32_t offset,
  uint8_t CANARD_MAYBE_UNUSED(root_item))
{
    uint32_t c = 0;
#ifndef CANARD_USE_FLOAT16_CAST
    uint16_t tmp_float = 0;
#else
    CANARD_USE_FLOAT16_CAST tmp_float = 0;
#endif

    // Compound
    offset = uavcan_Timestamp_encode_internal(&source->timestamp, msg_buf, offset, 0);

    // Compound
    offset = uavcan_Timestamp_encode_internal(&source->gnss_timestamp, msg_buf, offset, 0);
    source->gnss_time_standard = CANARD_INTERNAL_SATURATE_UNSIGNED(source->gnss_time_standard, 7)
    canardEncodeScalar(msg_buf, offset, 3, (void*)&source->gnss_time_standard); // 7
    offset += 3;

    // Void5
    offset += 5;
    canardEncodeScalar(msg_buf, offset, 8, (void*)&source->num_leap_seconds); // 255
    offset += 8;

    source->longitude_deg_1e8 = CANARD_INTERNAL_SATURATE(source->longitude_deg_1e8, 68719476735)
    canardEncodeScalar(msg_buf, offset, 37, (void*)&source->longitude_deg_1e8); // 68719476735
    offset += 37;

    source->latitude_deg_1e8 = CANARD_INTERNAL_SATURATE(source->latitude_deg_1e8, 68719476735)
    canardEncodeScalar(msg_buf, offset, 37, (void*)&source->latitude_deg_1e8); // 68719476735
    offset += 37;

    source->height_ellipsoid_mm = CANARD_INTERNAL_SATURATE(source->height_ellipsoid_mm, 67108863)
    canardEncodeScalar(msg_buf, offset, 27, (void*)&source->height_ellipsoid_mm); // 67108863
    offset += 27;

    source->height_msl_mm = CANARD_INTERNAL_SATURATE(source->height_msl_mm, 67108863)
    canardEncodeScalar(msg_buf, offset, 27, (void*)&source->height_msl_mm); // 67108863
    offset += 27;

    // Static array (ned_velocity)
    for (c = 0; c < 3; c++)
    {
		**tmp_float = canardConvertNativeFloatToFloat16(source->ned_velocity[c]);
        canardEncodeScalar(msg_buf, offset, 16, (void*)&tmp_float); // 32767**
        offset += 16;
    }

    source->sats_used = CANARD_INTERNAL_SATURATE_UNSIGNED(source->sats_used, 63)
    canardEncodeScalar(msg_buf, offset, 6, (void*)&source->sats_used); // 63
    offset += 6;

    source->status = CANARD_INTERNAL_SATURATE_UNSIGNED(source->status, 3)
    canardEncodeScalar(msg_buf, offset, 2, (void*)&source->status); // 3
    offset += 2;

    // float16 special handling
#ifndef CANARD_USE_FLOAT16_CAST
    tmp_float = canardConvertNativeFloatToFloat16(source->pdop);
#else
    tmp_float = (CANARD_USE_FLOAT16_CAST)source->pdop;
#endif
    canardEncodeScalar(msg_buf, offset, 16, (void*)&tmp_float); // 32767
    offset += 16;

    // Void4
    offset += 4;

    // Dynamic Array (position_covariance)
    // - Add array length
    canardEncodeScalar(msg_buf, offset, 4, (void*)&source->position_covariance.len);
    offset += 4;

    // - Add array items
    for (c = 0; c < source->position_covariance.len; c++)
    {
        canardEncodeScalar(msg_buf,
                           offset,
                           16,
                           (void*)(source->position_covariance.data + c));// 32767
        offset += 16;
    }

    // Dynamic Array (velocity_covariance)
    if (! root_item)
    {
        // - Add array length
        canardEncodeScalar(msg_buf, offset, 4, (void*)&source->velocity_covariance.len);
        offset += 4;
    }

    // - Add array items
    for (c = 0; c < source->velocity_covariance.len; c++)
    {
        canardEncodeScalar(msg_buf,
                           offset,
                           16,
                           (void*)(source->velocity_covariance.data + c));// 32767
        offset += 16;
    }

    return offset;
}`

pavel-kirienko, 2019-05-28:
@aasmune may I ask you to verify this, please?

williamyoung0, 2019-05-28:
After I made changes to the code, the data was parsed normally. The following is the modification:
`	**tmp_float = canardConvertNativeFloatToFloat16(source->ned_velocity[c]);**
        canardEncodeScalar(msg_buf, offset, 16, (void*)&tmp_float); // 32767
        offset += 16;`

closed 2019-09-12 by pavel-kirienko