OpenCyphal/libcanard #135: Wrong code generated for dynamic array of bools

mprymek, 2019-09-03:
When I compile this DSDL file:

```
#
# DigitalInputsValues
#

uint8 start_input_id

bool[<=32] values
```

I got illegal decode code generated:

```
uint32_t automation_DigitalInputsValues_encode_internal(automation_DigitalInputsValues* source,
  void* msg_buf,
  uint32_t offset,
  uint8_t CANARD_MAYBE_UNUSED(root_item))
{
    uint32_t c = 0;

    canardEncodeScalar(msg_buf, offset, 8, (void*)&source->start_input_id); // 255
    offset += 8;

    // Dynamic Array (values)
    6
    //  - Add array length, last item, but bitlen < 8.
    canardEncodeScalar(msg_buf, offset, 6, (void*)&source->values.len);
    offset += 6;

    // - Add array items
    for (c = 0; c < source->values.len; c++)
    {
        canardEncodeScalar(msg_buf,
                           offset,
                           1,
                           (void*)(source->values.data + c));// 1
        offset += 1;
    }

    return offset;
}
```

Note the strange `6` in `Dynamic Array (values)`.

I'm sorry, I'm not familiar enough with the compiler code to make a PR.

pavel-kirienko, 2019-09-06:
Sorry, the compiler is currently broken. We have added the following warning to the README to make sure people are aware of it:

> WARNING: this code generation tool is not production-ready; do not use it unless you are feeling adventurous and willing to contribute. In a production setting consider writing the serialization code manually instead.

We should be working on a new compiler in the UAVCAN-v1.0 branch. If you need code generation, please use libuavcan instead.

mprymek, 2019-09-06:
Ok, thanks.

closed 2019-09-06 by mprymek