OpenCyphal/libcanard #138: Acceptance filter configuration on STM32 (bxCAN) for standard IDs

ppescher, 2020-01-14:
This conditional expression causes the wrong bit format to be used when you add a filter with a standard id and you don't care about the IDE bit:
https://github.com/UAVCAN/libcanard/blob/8ee343c4edae0e0e4e1c040852aa3d8430f7bf76/drivers/stm32/canard_stm32.c#L519

I know it may be an unusual case, as that would accept also a whole lot of extended ids, but still the code is not doing what it is asked for and not even accept the standard id that you specify, as the bits are shifted like for extended ids.

A possible solution is to replace the "if" condition above with only:
```
if (cfg->id & CANARD_CAN_FRAME_EFF)
```
That makes the id bits shifted the correct way.
And then add this "else" branch in the following "if":
```
if (cfg->mask & CANARD_CAN_FRAME_EFF)
{
    mask |= CANARD_STM32_CAN_RIR_IDE;
}
else
{
    id |= CANARD_STM32_CAN_RIR_IDE;
}
```
To cope with the weird logic of bxCAN acceptance filters.

pavel-kirienko, 2020-01-15:
I did not dig deep, but it does seem correct. Would you volunteer to send a pull request?

ppescher, 2020-01-15:
HI Pavel, I'm not using libcanard (yet) but I found this implementation of acceptance filters for STM32 and used the same logic in another project after reading the comments section. I was looking for an explanation of how filter masks worked, the . Then I discovered this issue, but only tested my own rewrite, although pretty similar. So I can't provide a PR right now, sorry.