OpenCyphal/libcanard #141: Removed dsdl generator in libcanard v0

jxltom, 2020-02-21:
In this commit https://github.com/UAVCAN/libcanard/commit/dd7e92e995658b1c06f73638bf86f6cf5268713f, the DSDL generator is removed. The generated packages are quite useful since we don't need to write the encode and decode method by ourselves anymore, which saves a lot of time. I'm curious why this is removed? THANKS!

pavel-kirienko, 2020-02-21:
The generator was a nice idea, but, not pointing any fingers, it could have been done better. See https://github.com/UAVCAN/libcanard/pull/137.

<sub>(maybe [the forum](http://forum.uavcan.org/) is a better place for asking questions?)</sub>

jxltom, 2020-02-21:
Thanks for reply.

As uavcan v0 will still be maintained until at least 2021, the lack of DSDL generator makes using libcanard v0 a little bit inconvenient.

I will try to use the deprecated dsdl generator in my own fork for a while and see what I can do for it. For now, it looks like the bugs mentioned in https://github.com/UAVCAN/libcanard/pull/137 are not hard to fix.

pavel-kirienko, 2020-02-21:
Okay. I think we, the core devs, are not going to dedicate time to this in the foreseeable future, and I don't think it's a critical issue strategically, so I am going to close it.

closed 2020-02-21 by pavel-kirienko