OpenCyphal/libcanard #154: canardRxAccept fails with only one subcription

hugo-a-garcia, 2020-10-24:
if only one subscription is available (eg hearbeat) then canardRxAccept() fails with 
 // No matching subscription.

but

it does not fail with 2 or more subscriptions

Can someone confirm this please.

pavel-kirienko, 2020-10-24:
There is no perfect software but what you are describing seems unlikely. Can you please share any details?

pat1, 2020-10-24:
If I am not wrong, some problems occurs if  CanardRxSubscription "object" is not on global scope or not static declared ...

pavel-kirienko, 2020-10-24:
Where's the borrow checker when you need it?

Sounds like a lifetime management problem. Your subscription instances shall live at least until you call `canardRxUnsubscribe`. This is weakly hinted at by this statement but perhaps it's unclear:

https://github.com/UAVCAN/libcanard/blob/bd188fb82a79d5a3aba89b9e1504c9b85a015509/libcanard/canard.h#L265

hugo-a-garcia, 2020-10-24:
Here is the code:

https://github.com/bluecorn/megaavr-can-multicast

If you comment out the dummy subscription (lines 60-66) then the "Hearbeat Recieved\n" (line 114) is not printed.

pavel-kirienko, 2020-10-24:
Yeah, you have a lifetime management problem in your code. Make your subscriptions static. The docs have been clarified in https://github.com/UAVCAN/libcanard/pull/155, thanks for raising this.

pavel-kirienko, 2020-10-24:
You have another problem of similar nature here, this is going to break terribly:

https://github.com/bluecorn/megaavr-can-multicast/blob/479ae485c2f7782fd9985a9605ea2a346235c245/megaavr-can-multicast.ino#L200-L207

closed 2020-10-24 by pavel-kirienko

hugo-a-garcia, 2020-10-24:
Thank ypiu