OpenCyphal/libcanard #158: Allow source ID = destination ID for service requests

TomDeRybel, 2020-11-16:
I have come across a detail (canard.c, rxTryParseFrame, L579) that causes an issue for my implementation. 

Conceptually, (and this may be silly), I have my main computational node be both a service provider AND a client to itself for those general services also shared by other nodes in the network.
- The idea is that the services run in their own threads, serving network nodes and the main node itself indiscriminately.
- The program flow of the actual main node application can then just do its thing, like any node would.

This does not work right now, while it does work for regular message transfers. After some sleuthing, I came across a note in the aforementioned file that states: "Per Specification, source cannot be the same as the destination". Below that is the offending check for source ID == destination ID.

Q: Is there a deep reason why, for services, the source ID may not be equal to the destination ID?
Q: Where in the specification is this stated? I can't seem to find it.

pavel-kirienko, 2020-11-16:
> Q: Is there a deep reason why, for services, the source ID may not be equal to the destination ID?

In section **2.1.1.1 Architecture** we say:

> A UAVCAN network is a decentralized peer network, where each peer (node) has a unique numeric identifier — node-ID —  ranging from 0 up to a transport-specific upper boundary which is guaranteed to be not less than 127

Having conflicting node-IDs on the network is incompatible with most lower-layer transport protocols, such as CAN (which dictates that no two nodes shall publish identical arbitration-ID with differing payloads) or UDP/IP (where it is generally required that IP addresses on a subnet are unique); hence the restriction. Considering that, a transport frame where the source equals destination cannot occur on a well-defined network by design.

Maybe you need the processes on your main node interoperate using a different solution, like UAVCAN/UDP over the loopback interface?

> Q: Where in the specification is this stated? I can't seem to find it.

Section **4.1.1.4 Route specifier**:

> A *unicast transfer* is a transfer that is addressed to a specific single node whose node-ID is not the same as that of the origin

TomDeRybel, 2020-11-18:
Thanks again for the fast and detailed reply!
After consideration, I've decided to do the, in retrospect, obvious thing and just trap these requests and respond directly, bypassing any networking. (Due to a design faux-pas, the service and client handing for a service type is all in the same class, and the client handling code thus has the service information readily available.)

Best regards,
Tom

closed 2020-11-18 by TomDeRybel