OpenCyphal/libcanard #161: Typo in canard.h, L582 provably -> probably

wucke13, 2021-01-15:
https://github.com/UAVCAN/libcanard/blob/55938c51cf7da3096679933302693c9dae7a2b33/libcanard/canard.h#L582

pavel-kirienko, 2021-01-15:
This is not a typo though. The point is that it can be proven that the behavior is invariant to the number of nodes on the network.

closed 2021-01-15 by pavel-kirienko

wucke13, 2021-01-15:
I dd not find this word in the English dictionary of my choice. To me, `verifiable` would be a better fit (even though it is broader in semantics). Sorry for the noise though, probably would made sense well enough for me to not consider other intentional wording.

pavel-kirienko, 2021-01-15:
English is not my first language so maybe the problem is there but I can't see it. @clyde-johnston can you please advise?

clyde-johnston, 2021-01-15:
The spelling and usage is fine: 'provably' is the adverb of 'provable'. The usage is unusual because two adverbs are used: 'identically' qualifies 'perform', and 'provably' qualifies 'perform identically'. The split infinitive is also problematic. Not knowing how it is proved or to whom, it is not clear (to me at least) why 'provably' is an important qualification. If the word were removed, it would appear to be equally valid. This all depends on the context.

wucke13, 2021-01-16:
I have no doubt now that it is formally correct, and it sounds eloquent for sure, and it is probably not that important too. That being said, I would imagine ideal documentation to worded using adequate word yet simple phrasing. The primary goal should be someone (who might not be a native speaker themselves) can read it, and understand it, preferable without an english refresher :smile: . Your wording decides to which folks you appeal. Considering that, this might not even be the wrong move, but certainly I will not be the last one to stumble over this.

@clyde-johnston comment regarding the qualification I do agree, there is few benefits in stating that something is possible to prove without mentioning any details about the nature and implications of said not further specified proof.