OpenCyphal/libcanard #166: Problems in porting libcanard to stm32f405RGTX

295951389, 2021-05-18:
I refer to the following website transplant code:[https://kb.zubax.com/display/MAINKB/1.+Basic+tutoria](https://kb.zubax.com/display/MAINKB/1.+Basic+tutoria) .The final result is that the data is sent into a dead loop:
What should I do when the final sending part enters here?

`        for (uint8_t i = 0; i < 3; i++)
        {
            if (tme[i])                                 // This TX mailbox is free, we can use it
            {
                tx_mailbox = i;
            }
/******What should I do when the final sending part enters here?*****/
            else                                        // This TX mailbox is pending, check for priority inversion
            {
                if (!isFramePriorityHigher(frame->id, convertFrameIDRegisterToCanard(BXCAN->TxMailbox[i].TIR)))
                {
                    // There's a mailbox whose priority is higher or equal the priority of the new frame.
                    return 0;                           // Priority inversion would occur! Reject transmission.
                }
            }
        }`

pavel-kirienko, 2021-05-18:
Retry transmission later. This is part of the inner priority inversion avoidance algorithm, described in UAVCAN Specification, section [**4.2.4.3 Inner priority inversion**](https://uavcan.org/specification#page=70).

closed 2021-05-18 by pavel-kirienko

295951389, 2021-05-18:
Why frame > ID and  convertFrameIDRegisterToCanard (BXCAN > TxMailbox [i]. TIR)
Always equal, no matter how long I wait?

pavel-kirienko, 2021-05-18:
This means that the CAN controller can't deliver the pending frame to the bus. Maybe the bus is malfunctioning or there are no other nodes on it to confirm reception of the frame.

295951389, 2021-05-19:
> This means that the CAN controller can't deliver the pending frame to the bus. Maybe the bus is malfunctioning or there are no other nodes on it to confirm reception of the frame.

I have two boards, one of which is broken, and the second PCB is normal,Thanks.