OpenCyphal/libcanard #172: Travis CI is no more, migrate to GitHub Actions

pavel-kirienko, 2021-07-20:
https://github.com/UAVCAN/libcanard/blob/master/.travis.yml

pavel-kirienko, 2021-09-14:
Reopening because there are still configuration issues with CI.

- [ ] Test coverage measurement is not working, the reported coverage is 0%. This is probably because the sonar scanner configuration is incomplete -- it used to include more options:

https://github.com/UAVCAN/libcanard/blob/2a116170285fb47fcaae150ad21c2ccde0756a5f/.travis.yml#L33-L39

- [ ] SonarQube is analyzing irrelevant files, specifically `catch.hpp` (the unit testing framework) and CMake-generated C++ files (because the `sonar.sources` parameter is configured incorrectly):
   [image]

Please avoid merging things without review, as it would help us avoid issues like this.

coderkalyan, 2021-09-14:
> Please avoid merging things without review, as it would help us avoid issues like this.

Sorry, my intent was not to merge without review, but I found that the sonarcloud was not actually running anything on the PR branch since no code had changed. Now that I know it runs correctly, I will fix these issues and submit for review.

pavel-kirienko, 2021-09-18:
Both of these deficiencies have been addressed by #181, thanks! One new issue is that Travis-CI integration is still enabled for some reason, as I wrote in a comment there, so I am going to keep this open.

closed 2021-10-02 by coderkalyan