OpenCyphal/libcanard #178: build check error: narrowing conversion from 'int' to signed type 'int8_t' (aka 'signed char') is implementation-defined

Linjieqiang, 2021-08-12:
[image]

pavel-kirienko, 2021-08-12:
I don't see where the conversion is taking place. Which version of Clang-Tidy is this? Libcanard is currently analyzed with Clang Tools version 11.

Linjieqiang, 2021-08-12:
I use the segger embedded studio ide to analyze the code. The wiki: https://wiki.segger.com/Static_code_analysis_in_Embedded_Studio

pavel-kirienko, 2021-08-12:
Idk this doesn't make sense to me. Can you see what it takes to fix this warning? Maybe I am missing something.

Linjieqiang, 2021-08-12:
[image]

Linjieqiang, 2021-08-12:
https://clang.llvm.org/extra/clang-tidy/checks/cppcoreguidelines-narrowing-conversions.html

Linjieqiang, 2021-08-12:
`(int8_t)((-(int8_t)(uint8_t) ~val) - 1)` and `((int8_t) val)`

Linjieqiang, 2021-08-12:
When I remove the `libcanard\.clang-tidy`, the code static check is fine.

Linjieqiang, 2021-08-12:
Maybe you could use the clang 12 to try CI?

thirtytwobits, 2021-08-12:
Hmm. I'm not able to reproduce. Can you take a look here:

https://godbolt.org/z/Mdrar3jTo

Linjieqiang, 2021-08-16:
That is really a strange problem, I think it has nothing to do with IDE. But why this error is reported, I am not sure.

Linjieqiang, 2021-08-16:
Maybe it was a warning, but the inspection conditions are too strict. @thirtytwobits

pavel-kirienko, 2021-08-16:
@Linjieqiang it would help if you could isolate the issue to a minimal working example. Also, consider checking the Clang bug tracker to see if this is a known issue. Maybe ask the maintainers about this for good measure unless someone did that already. In the worst case, we may need to add a NOLINT comment on the offending lines to suppress this diagnostic.

Linjieqiang, 2021-08-19:
Still haven't found the reason, why don't we use NOLINT to circumvent this problem?

pavel-kirienko, 2021-08-20:
We need an MWE first. Please extract the offending fragment into a separate unit and see if you can reproduce the problem there.

closed 2021-12-04 by pavel-kirienko

Linjieqiang, 2021-12-05:
Thanks. @pavel-kirienko

pavel-kirienko, 2021-12-05:
I didn't really fix it, the code in question was removed from the library (use Nunavut instead).

PR #182 (2021-12-04): Libcanard v2.0
- Closes #168 
- Fixes #175 
- Closes #176 -- while the demos that are shown in the README still don't use Nunavut, they are explicit about it.
- Closes #178 by removing the offending code (use Nunavut instead)

See the changelog in the README for a brief overview of the changes. The most interesting changes are two:

- Split API into smaller entities to improve its orthogonality and fix the problem of const correctness reported by Scott.
- Replace all linked lists with AVL trees to reduce the asymptotic complexity of both RX and TX pipelines from linear to logarithmic.

@coderk [... cut]