OpenCyphal/libcanard #184: Some platforms have no uint8_t etc. (one should use uint8_least_t/uint8_least_t?)

TYuD, 2021-11-20:
Hi!

I see type uint8_t in your code, but some DSPs and MCUs have size of byte bigger then 8 bit. There are examples of stdint.h file (C/C++):

// -------------------------------------- ADSP TS-201 compiler:

typedef unsigned int uint32_t;

typedef uint32_t uint_least8_t;
typedef uint32_t uint_least16_t;
typedef uint32_t uint_least32_t;

typedef uint32_t uint_fast8_t;
typedef uint32_t uint_fast16_t;
typedef uint32_t uint_fast32_t;

// -------------------------------------- TMS320C2804 compiler:

typedef unsigned int uint16_t;
typedef unsigned long uint32_t;

typedef uint16_t uint_least8_t;
typedef uint16_t uint_least16_t;
typedef uint32_t uint_least32_t;

typedef uint16_t uint_fast8_t;
typedef uint16_t uint_fast16_t;
typedef uint32_t uint_fast32_t;

I think you should use uint8_least_t/uint8_least_t instead of uint8_t etc.

I'm wrong?

pavel-kirienko, 2021-11-20:
I don't think you can easily replace uint8 as it would likely break transfer payload management, which assumes that the buffer memory is byte-addressable. If you experimented with it and reported back your findings, I think that would be useful to many users, as similar issues were raised in the past (e.g., https://github.com/UAVCAN/libcanard/issues/55).

TYuD, 2021-11-20:
Sorry. I liked your startup. It is much demanded. But I can't use it.

I use bitfields in structures and bitshifting to pack data:

//----------------------------------------------------
typedef struct
{
  uint_fast16_t TypeFormat : 16;
  uint_fast8_t  SystemFlags: 8 ;
  uint_fast8_t  QueryID    : 8 ;
  uint_fast32_t Reserve    : 32;
}
TPressureSensorQuery;
//----------------------------------------------------
typedef struct
{
  uint_fast16_t TypeFormat : 16;

  uint_fast8_t  Error      : 1 ;
  uint_fast8_t  Ready      : 1 ;
  uint_fast8_t  SystemFlags: 6 ;
  uint_fast8_t  UserFlags  : 8 ;

  uint_fast32_t Pressure   : 32;

  uint_fast32_t Time       : 32;
  uint_fast32_t RareAuxData: 32;
  uint_fast8_t  RareAuxID  : 8 ;
  uint_fast32_t Reserve    : 24;
  uint_fast32_t AuxData1   : 32;
  uint_fast32_t AuxData2   : 32;
}
TPressureSensorAnswer;

TYuD, 2021-11-20:
Can you highlight typical sections of code where the discussed problem brings difficulties? Together we could think about how to solve them. It would be easier for me if these sections were in C/C++ language.

pavel-kirienko, 2021-11-20:
I would recommend you to approach it this way instead: modify the codebase to use `uint_least8_t` in place of `uint8_t` and see if that causes breakage. You will need to modify the test suite to run on your platform though, which may prove tedious, so you should start with the most basic tests that validate the fundamentals, like https://github.com/UAVCAN/libcanard/blob/master/tests/test_private_rx.cpp.

TYuD, 2021-11-20:
It is difficult for me to understand much amount of your code. 

Types like uint8_fast_t I use for a long time. There are no problems with them as variables or structure fields. 

Problems can be with pointers. They can not offset to 8 bits (uint8_t) in some platforms. In such a case one must use bit operations to pack/unpack data instead of pointer offset. I think you do just so when pack uintX_t (X<8, DSDL) into a message. In same a way data may be packet into char array, when size of char is bigger 8 bit. Char size may be parameterized in your code. Its value contain CHAR_BIT macro in file limits.h. Also this value can be pointed as a parameter of DSDL translators like little/big endean. It may be considered like a MTU value in transport layer.

TYuD, 2022-01-09:
Hi, Pavel!

I have slightly modified the nunavutCopyBits function (I think this is the core function of serilization). Now it looks the nunavutCopyBits can work on platforms with different byte lengths. Even a little faster (without divisions). In the same way, one can modify the other functions of the library.

It is here: 
`
#include <string.h>
#include <limits.h>

static inline size_t nunavutChooseMin(const size_t a, const size_t b)
{
    return (a < b) ? a : b;
}
// ----------------------------------------------------
#if CHAR_BIT == 8
  #define CHAR_SHIFT 3
#elif CHAR_BIT == 16
  #define CHAR_SHIFT 4
#elif CHAR_BIT == 32
  #define CHAR_SHIFT 5
#elif CHAR_BIT == 64
  #define CHAR_SHIFT 6
#else
  #error Strange CHAR_BIT value
#endif

#define CHAR_SHIFT_MASK  ((1u << CHAR_SHIFT) - 1u)
// ----------------------------------------------------
static inline void nunavutCopyBits(void* const dst, const size_t dst_offset_bits, const size_t length_bits, const void* const src, const size_t src_offset_bits)
{
    unsigned ch_shift = CHAR_SHIFT;
    unsigned ch_mask  = UCHAR_MAX ;
    if (0 == (src_offset_bits & CHAR_SHIFT_MASK) && 0 == (dst_offset_bits & CHAR_SHIFT_MASK))  // Aligned copy, optimized, most common case.
    {
        const size_t length_chars = size_t(length_bits >> CHAR_SHIFT);
        const char* const psrc    = (src_offset_bits >> CHAR_SHIFT) + (const char*)src;
        char*       const pdst    = (dst_offset_bits >> CHAR_SHIFT) + (      char*)dst;
        memmove(pdst, psrc, length_chars);
        const unsigned length_mod = length_bits & CHAR_SHIFT_MASK;
        if (length_mod == 0)
            return;
        const char* const last_src = psrc + length_chars;
        char* const       last_dst = pdst + length_chars;
        const unsigned    mask     = (1u << length_mod) - 1;
        *last_dst = (*last_dst & unsigned(~mask)) | (*last_src & mask);
        return;
    }
    const char* const psrc     = (const char*)src;
    char*       const pdst     = (      char*)dst;
    size_t            src_off  = src_offset_bits;
    size_t            dst_off  = dst_offset_bits;
    const size_t      last_bit = src_off + length_bits;
    while (last_bit > src_off)
    {
        const unsigned src_mod = src_off & CHAR_SHIFT_MASK;
        const unsigned dst_mod = dst_off & CHAR_SHIFT_MASK;
        const unsigned max_mod = src_mod > dst_mod ? src_mod : dst_mod;
        const unsigned size    = nunavutChooseMin(CHAR_BIT - max_mod, last_bit - src_off);
        const unsigned mask    = (((1u << size) - 1) << dst_mod) & UCHAR_MAX;
        const unsigned in      = (unsigned(psrc[src_off >> CHAR_SHIFT] >> src_mod) << dst_mod) & UCHAR_MAX;
        const unsigned a       = pdst[dst_off >> CHAR_SHIFT] & (~mask);
        const unsigned b       = in & mask;
        pdst[dst_off >> CHAR_SHIFT] = a | b;
        src_off += size;
        dst_off += size;
    }
}
`
Here are some small checks on the PC and on the TigerSHARC processor:

`void main()
{
  #if  CHAR_BIT == 8

    const size_t N = 6;

    char src[N] = { '\x00','\xA0','\xBC','\x0D','\x00','\x00' };
    char dst[N] = { '\xFF','\xFF','\xFF','\xFF','\xFF','\xFF' };
    char res[N] = { '\xAF','\xBC','\xFD','\xFF','\xFF','\xFF' };

    const size_t dst_offset_bits = 4 ;
    const size_t length_bits     = 16;
    const size_t src_offset_bits = 12;

  #elif CHAR_BIT == 32

    const size_t N = 4;

    char src[N] = { '\x00000000','\xA0000000','\xBCDE1234','\x00000000' };
    char dst[N] = { '\xFFFFFFFF','\xFFFFFFFF','\xFFFFFFFF','\xFFFFFFFF' };
    char res[N] = { '\xDE1234AF','\xFFFFFFBC','\xFFFFFFFF','\xFFFFFFFF' };

    const size_t dst_offset_bits = 4 ;
    const size_t length_bits     = 36;
    const size_t src_offset_bits = 60;

  #else
    #error Strange CHAR_BIT value
  #endif

  nunavutCopyBits(dst, dst_offset_bits, length_bits, src, src_offset_bits);
  bool Ok = true;
  for( unsigned i=0; i < N; i++ )
    if( dst[i] != res[i] )
      Ok = false;
}
`

pavel-kirienko, 2022-01-09:
I opened a new issue over at Nunavut to track this: https://github.com/UAVCAN/nunavut/issues/233

>Even a little faster (without divisions)

With any half-decent compiler, replacing divisions with shifts adds nothing but obscurity.