OpenCyphal/libcanard #191: Examples required

LightningPORTO, 2022-01-28:
Hello,

For context, I'm attempting to develop a UAVcan implementation using cannard, on a STM32 using RTOS.

I find the examples on the readme satisfactory and I believe we could benefit from full a example of publishing and subscribing to messages and services.
I don't mean specifically for my case (would also be nice), but a very generic (maybe heartbeat publish and subscribe?) implementation, fully written on a C example.

Could this be arranged? If there is anything I haven't found using the current version found on this repo.

Thank you for this amazing protocol and amazing work.

pavel-kirienko, 2022-01-28:
Thanks for the kind words.

We have full-featured demos here, but they are for an earlier revision of this library where the API is slightly different: https://github.com/UAVCAN/demos

The full API diff can be seen here (scroll to `canard.h` and ignore the rest):

https://github.com/UAVCAN/libcanard/commit/cf0a8bc67ddf98dfccf157695533ac5324e021eb#diff-6d1addc3815ef077e06339f13578aa999d57eba65be6a707e2fe55d4e0038509

LightningPORTO, 2022-01-28:
Thank you for such a quick answer Pavel,

I was hoping I could get a barebones example of a publish and subscribe to a heartbeat message for example.
The given examples are great but very complex ( at least to me ) to understand the steps required to implement such basic functions.

When I mean "barebones" I'm actually trying to get little increments at a time, the protocol is great but with a very steep learning curve. For example, ROS has tutorials for publishing and subscribing... 

I'm not saying to get a full wiki and all going on, but maybe a little beginner-friendly approach wouldn't hurt.

Maybe a simple (and possibly separate examples) to publish and subscribe to the heartbeat messages would be a good start?

Once again, there are just suggestions, but coming from someone who has average knowledge in programing and really sees the potential of this protocol.

silverv, 2022-01-31:
While not exactly a bare bones example, this Sapog repository is closer to what you need than the demo repository:
- it runs on actual hardware instead of just as a process on a computer
- uses Canard v2
- is supposed to become a reference project for new adopters
See that first. I haven't planned on when I will be making bare bones examples yet.

LightningPORTO, 2022-02-01:
> 

This is somewhat closer to what I was looking for. Im attempting to implement libcannard to talk to a PX4 autopilot, using a Nucleo board STM. I know there are examples out there, but none are using the latest libcannard...

Thank you for your sugestion, if in the future you do a barebones, please let me know.

coderkalyan, 2022-03-13:
I think I have an example using Zephyr RTOS somewhere, but its also using the canard V1 API. If I find some time I'll update it and share.

ValeriiKim, 2022-04-09:
> This is somewhat closer to what I was looking for. Im attempting to implement libcannard to talk to a PX4 autopilot, using a Nucleo board STM. I know there are examples out there, but none are using [...]

If you are interested here is my repo where libcanard v2 is used: https://github.com/ValeriiKim/Modrob-UAVCAN/blob/master/src/SENSOR_MODULE/main.cpp

pavel-kirienko, 2022-05-05:
The demos have been updated to use the new libcanard v3: https://github.com/OpenCyphal-Garage/demos/pull/12

Dmivaka, 2022-09-06:
> This is somewhat closer to what I was looking for. Im attempting to implement libcannard to talk to a PX4 autopilot, using a Nucleo board STM. I know there are examples out there, but none are using [...]

I don't know if it's fair to call the use of STM32Cube "barebone", but here is a simple Libcanard 2 example for it, with project generation from .ioc - https://github.com/Dmivaka/STM32-HAL-Libcanard

githubfa, 2022-09-24:
> 

one simple demo is  better than 1000 words😂.  i am a beginner of uavcan, and it's difficult for me to find out  such as: how the demo work, what is the relation ship between libcanard and DSDL? how can i use  libcanard  with DSDL?  it's  all  question ...

pavel-kirienko, 2022-09-24:
The demos I linked earlier illustrate both. Take a look at this one: https://github.com/OpenCyphal-Garage/demos/tree/main/differential_pressure_sensor

It is a very basic application that incorporates both libcanard and DSDL code generation (via Nunavut). You can easily port it to an embedded platform (it can be run on GNU/Linux out of the box).

You can also check out @aentinger's [Arduino Cyphal](https://github.com/107-systems/107-Arduino-Cyphal) and its examples -- it might be more approachable.

githubfa, 2022-09-27:
> The demos I linked earlier illustrate both. Take a look at this one: https://github.com/OpenCyphal-Garage/demos/tree/main/differential_pressure_sensor
>
> It is a very basic application that [...]

Thanks a lot！ maybe those demo can be join into the README.md to give other more help, even if it's just a link.

mehrkonfetti, 2022-10-10:
Would you still be interested in adding more minimal samples (like a heartbeat subscriber & publiser) to eg [the demos repository](https://github.com/OpenCyphal-Garage/demos/)? I have found myself looking for something like this a couple of times now and would love to implement it.

pavel-kirienko, 2022-10-10:
>Would you still be interested in adding more minimal samples (like a heartbeat subscriber & publiser) to eg [the demos repository](https://github.com/OpenCyphal-Garage/demos/)?

Yes.

>I have found myself looking for something like this a couple of times now and would love to implement it.

Please do. As an alternative to the demos repo, you can also publish a link to your example here: https://forum.opencyphal.org/t/libcanard-examples-starters-tutorials/935

pavel-kirienko, 2025-04-26:
We have updated demos published in the demos repository so I am closing it. More demos are always welcome.

closed 2025-04-26 by pavel-kirienko