OpenCyphal/libcanard #201: The tests enable -Werror but generate warnings

joshvazquez, 2022-10-12:
The tests enable -Werror here: https://github.com/OpenCyphal/libcanard/blob/master/tests/CMakeLists.txt#L39

The tests also generate warnings, such as unused variable _ [here](https://github.com/OpenCyphal/libcanard/blob/master/tests/helpers.hpp#L76), conversion to long unsigned int from const int may change the sign of the result [here](https://github.com/OpenCyphal/libcanard/blob/master/tests/helpers.hpp#L274). So the tests fail with warnings as errors.

Is this expected? I am using libcanard in CI/CD and would like to keep the tests enabled.

pavel-kirienko, 2022-10-12:
This is not expected. The code should be updated to eliminate the warnings you listed. A pull request would be appreciated.

joshvazquez, 2022-10-12:
Can't explain this.

Sign conversion warning:
```
const auto new_size = size_before + static_cast<std::size_t>(ret); // <-- here
enforce((ret < 0) || (new_size == que_.size),
    "Unexpected size change after push");
```

OK:
```
const auto num_added = static_cast<std::size_t>(ret);
enforce((ret < 0) || (size_before + num_added == que_.size),
    "Unexpected size change after push");
```
Compiler magic?

pavel-kirienko, 2022-10-12:
Evidently, integer promotion changes the type of the result, which affects the warning.

joshvazquez, 2022-10-12:
I'll submit a PR tomorrow on my work account. @joshvazquez-amzn