OpenCyphal/libcanard #203: -fsanitize=undefined runtime error: member access within null pointer of type 'struct TxItem'

jvishnefske, 2022-10-28:
I compiled canard.c from release tarball 3.0.0, and the public catch2 tests with my meson file. -fsanitize=undefined seems to be triggering while running unit tests.

../subprojects/libcanard-3.0.0/tests/test_public_tx.cpp:101: 
PASSED:
  REQUIRE( 3 == alloc.getNumAllocatedFragments() )
with expansion:
  3 == 3
[image]

../subprojects/libcanard-3.0.0/libcanard/canard.c:413:47: runtime error: member access within null pointer of type 'struct TxItem'
../subprojects/libcanard-3.0.0/tests/test_public_tx.cpp:107: 
PASSED:
  REQUIRE( -3 == que.push(&ins.getInstance(), 1'000'000'000'300ULL, meta, 100, payload.data()) )
with expansion:
  -3 == -3

```
#meson setup --warnlevel 2 --werror -Db_lto=false -Db_coverage=true -Db_sanitize="undefined" build-dir
project('canard', ['c','cpp'],
  version : '3.0.0',
  default_options : ['cpp_std=c++17'])

canard_incdir = include_directories('libcanard', '.', is_system : false)
canard_libsources = files('libcanard/canard.c')
canard_lib = library('canard', canard_libsources)
canard_dep = declare_dependency(
    link_with: canard_lib,
    include_directories : canard_incdir)

test_sources_public = [
'tests/test_public_tx.cpp',
'tests/catch/main.cpp',
'tests/test_public_roundtrip.cpp',
'tests/test_self.cpp',
'tests/test_public_filters.cpp',
'tests/test_public_rx.cpp',
]
# build subset which tests public interface.
canard_interface_tests = executable(
        'canard_test',
        test_sources_public,
        dependencies:[canard_dep, dependency('threads')],
        include_directories:['tests/catch','tests'],
        install: false,

    )
test('canard_tests', canard_interface_tests)
```

pavel-kirienko, 2022-10-29:
Thanks! The first one is legit -- although it has no practical significance, it is technically a UB, so I opened #204 to fix it. The other two are unhelpful.

Could you please help us set up the sanitizer in the CI?

closed 2022-10-30 by pavel-kirienko

jvishnefske, 2022-11-01:
Started #205 in attempt to add sanitizer to cmake build. It may take some experimenting to get right on CI.

PR #204 (2022-10-30): Fix 203
Fixes #203
commit 989124d0 (2022-10-30): Fix 203 (#204)

* Fix #203

* Update the changelog