OpenCyphal/libcanard #222: Extend api with `canardRxGetSubscription`

serges147, 2024-04-15:
c++ libcyphal needs api to detect already existing subscription (without "touching" one if any).
Here is proposal:
```
/// The return value is 1 if such subscription existed.
/// The return value is 0 if such subscription does not exist.
/// The return value is a negated invalid argument error if any of the input arguments are invalid.
int8_t canardRxGetSubscription(CanardInstance* const       ins,
                               const CanardTransferKind    transfer_kind,
                               const CanardPortID          port_id,
                               const size_t                extent,
                               CanardRxSubscription* const out_subscription);
```

serges147, 2024-04-22:
Looks like `canardRxHasSubscription() → bool` is enough - we only need to check for possible double subscription for the same kind & port pair.

pavel-kirienko, 2024-04-22:
I actually wanted to mention this earlier but forgot: since we'll have to find that specific subscription in the tree anyway, should we not keep `canardRxGetSubscription` as it is more versatile? There are sensible use cases where one might want to find an existing subscription to modify it.

serges147, 2024-05-06:
ok, I revert back to → `canardRxGetSubscription`

closed 2024-05-06 by serges147