OpenCyphal/libcanard #227: Add to ESP Component Registry

manforowicz, 2024-08-29:
What do you think about adding libcanard to the [ESP Component Registry](https://components.espressif.com/)?
Since the ESP32 has great CAN bus support, having libcanard packaged in an ESP component would make it more accessible to developers.

Doing this would probably entail:
1. Requesting an OpenCyphal namespace through the [ESP Registry permissions](https://components.espressif.com/settings/permissions/) page. Giving key contributors access to this namespace.

2. Packaging up libcanard by following the [Packaging ESP-IDF Components](https://docs.espressif.com/projects/idf-component-manager/en/latest/guides/packaging_components.html) guide. This requires adding some metadata and build files.

3. Uploading the package to the ESP registry each time there's an update.

pavel-kirienko, 2024-08-29:
Sounds interesting and I will support whatever activities are needed

kumekay, 2024-09-02:
Hello @Manforowicz, I'm Sergei from Espressif, thank you so much for sending your request for namespace `opencyphal` in the ESP Component Registry. Since you are not an active contributor to the project, before approving your request, I decided to discuss the topic here.

@pavel-kirienko if you want a full control over components in the ESP Component Registry, could you please sign up here: https://components.espressif.com/users/login? Then you will be able to give permissions to other developers.

BTW, for ease of uploading I would highly recommend using our GitHub Action https://github.com/espressif/upload-components-ci-action 

@XDanielPaul FYI

pavel-kirienko, 2024-09-03:
Thank you @kumekay for the advice. I have attempted to request the opencyphal namespace but it's already taken, presumably by @manforowicz. Could you please transfer it to me as the owner and then I will share access with the other team members? Thank you!

[image]

pavel-kirienko, 2024-09-03:
I also requested `cyphal` for security reasons.

XDanielPaul, 2024-09-03:
Hello @pavel-kirienko, I'm Daniel from Espressif. 
I have approved your namespace request for the namespace `cyphal` and would further want to ask you to request the namespace `opencyphal` once again in the ESP Component Registry, it should be available now :)

pavel-kirienko, 2024-09-03:
Thank you @XDanielPaul, this has been done.

@thirtytwobits FYI -- you don't seem to be registered in the ESP registry yet

pavel-kirienko, 2024-09-03:
@manforowicz a pull request amending the GitHub actions pipelines as necessary would be accepted.

XDanielPaul, 2024-09-03:
Thank you so much @pavel-kirienko, I have approved your namespace request for the `opencyphal` as well. You should be able to add other users now

PR #229 (2024-09-18): ESP-IDF Component Continuous Integration
This pull request is meant to resolve #227 by adding continuous integration for publishing Libcanard to the [ESP Component Registry](https://components.espressif.com/).

Files added:
- `idf_component.yml`. This file contains required ESP component metadata.
- `Kconfig`. ESP components use this file as a standard way for controlling build options. I've added options for enabling/disabling assertions and the CRC table.
- `CMakeLists.txt`. This file is required to register the component with the ESP build system.
- `.github/workflows/esp_publish.yml`. This [... cut]
commit 551af7fc (2024-09-18): ESP-IDF Component Continuous Integration (#229)

This pull request is meant to resolve #227 by adding continuous integration for publishing Libcanard to the [ESP Component Registry](https://components.espressif.com/).

Files added:
- `idf_component.yml`. This file contains required ESP component metadata.
- `Kconfig`. ESP components use this file as a standard way for controlling build options. I've added options for enabling/disabling assertions and the CRC table.
- `CMakeLists.txt`. This file is required to register the component with the ESP build system.
- `.github/workflows/esp_publish.ym [... cut]
closed 2024-09-18 by pavel-kirienko