OpenCyphal/libcanard #241: Heartbeat Serialization -- Difference between libcanard and PyCyphal

Scoeerg, 2025-03-29:
Hi all :) 

I am currently working with both libcanard and PyCyphal. Given two minimal example publishers on the same CAN-bus, I get different candumps:

```bash
//PyCyphal Heartbeat Minimal Publisher
can0  107D556A   [8]  01 00 00 00 00 00 38 E1
can0  107D556A   [8]  02 00 00 00 00 00 38 E2
can0  107D556A   [8]  03 00 00 00 00 00 38 E3
can0  107D556A   [8]  04 00 00 00 00 00 38 E4
```

with extended CAN ID (see pages 58, 59 in the [specifications.pdf](https://opencyphal.org/specification/Cyphal_Specification.pdf))

```bash
107D556A
00010000 01111101 01010101 01101010
// human-readable:
bits 0 to 6:
Node-ID = 106
bit 7:
reserved =  0
bits 8 to 20:
subject-ID = 7509
bits 21-23:
reserved = 011
bit 24:
anonymous = 0 //so NOT anonymous
bit 25:
service = 0 //message, NOT a service
bits 26-28:
priority = 4 //default 
```

and 

```bash
//libcanard Heartbeat Minimal Publisher
can0  107D550A   [7]  01 00 00 00 00 00 2A
can0  107D550A   [7]  02 00 00 00 00 00 2A
can0  107D550A   [7]  03 00 00 00 00 00 2A
can0  107D550A   [7]  04 00 00 00 00 00 2A
can0  107D550A   [7]  05 00 00 00 00 00 2A
```

```bash
107D550A
00010000 01111101 01010101 00001010
// human-readable:
bits 0 to 6:
Node-ID = 10
bit 7:
reserved =  0
bits 8 to 20:
subject-ID = 7509
bits 21-23:
reserved = 011
bit 24:
anonymous = 0 //so NOT anonymous
bit 25:
service = 0 //message, NOT a service
bits 26-28:
priority = 4 //default 
```

where the `PyCyphal` Heartbeat is actually registered by [Yakut Monitor](https://github.com/OpenCyphal/yakut) correctly, but the `libcanard` node is not. 

Now, the obvious difference between the two messages is only in it's body (and the different Node IDs of 106 and 10 in the extended CAN ID, but let's neglect that):

1) the length of the message body is 8Byte in the PyCyphal and 7Byte in the libcanard case
2) the "additional" 1byte in PyCyphal is changing over time `E1`, `E2` ... and seems to count upwards. 

 Now, if we look into the [public_regulated_data_types heartbeat message](https://github.com/OpenCyphal/public_regulated_data_types/blob/master/uavcan/node/7509.Heartbeat.1.0.dsdl):

```c
uint32 uptime                       
Health.1.0 health //de facto uint8_t
Mode.1.0 mode //de facto uint8_t
uint8 vendor_specific_status_code
```

and then into the `c`-generated code for the heartbeat message gives me:

```c
uavcan_node_Heartbeat_1_0 hbeat_message = {.uptime =some_function_to_measure_uptime_in_seconds(), //32bit
                                               .health = uavcan_node_Health_1_0_NOMINAL, //8bit
                                               .mode   = uavcan_node_Mode_1_0_OPERATIONAL, // 8bit
                                               .vendor_specific_status_code = vendor_specific_code //8bit
```

the body is supposed to be 7Byte (56bit), which luckily is also the `uavcan_node_Heartbeat_1_0_SERIALIZATION_BUFFER_SIZE_BYTES_` specified. Where does the discrepancy come from? Why does the PyCyphal Heartbeat have 8Byte? What's the meaning of `E1`, `E2` ... that is not part of the specification?

pavel-kirienko, 2025-03-29:
You forgot to increment the transfer ID

Scoeerg, 2025-03-29:
Oh for FUCKS SAKE. Thanks @pavel-kirienko

closed 2025-03-29 by Scoeerg